fair-acc/pulsed-power-ml #81: [20pt] Implementation, training and model validation using the approach from Sidhant Gupta

ThomasKittler, 2022-10-21:
[Sidhant Gupta, "ElectriSense: Single-Point Sensing using EMI for Electrical Event
Detection and Classification [..]",PhD thesis, U-Washington, 2014](http://hdl.handle.net/1773/26334)

mariahirsch, 2022-11-22:
Labelling training data more difficult than anticipated: there is no threshold value for peaks in Apparent Power Spectra that works as a switch detection for all appliances. Even the sum of the individual spectrum of a single time step is not comparable, although switches are clearly to be seen. 

This are the sums of a time series of switches of a blow dryer:

[image]

And the sums of a time series of switches of an LED bulb:

[image]

Note the difference on the y axis.

mariahirsch, 2022-11-22:
New approach: norm spectra before switch detection via sum of spectrum. Examples: 

LED bulb:

[image]

or 

blow dryer:

[image]

For switch detection: Once any switch event is detected for a normed, background subtracted spectrum, the detection of the switch  direction (on or off) is preferrably done with the <switching_offset>th next spectrum. This approach is more robust against peaks resulting from mechanical switches.

closed 2022-12-16 by RalphSteinhagen